eugene-tarassov/vivado-risc-v #242: VCU118 boot failed with v3.6.0

Yuxin-Yu, 2024-07-24:
I used vcu118 with pmod SD card module to start Linux. The relevant Block Design, tcl script, and dts files are shown below
 
[image]
```
set_property -dict [ list \
    CONFIG.sdio_card_detect_level {0} \
    CONFIG.capabilies_reg {0x0001} \
  ] $SD
```
```
 mmc: mmc0@60000000 {
            compatible = "riscv,axi-sd-card-1.0";
            clock = <100000000>;
            reg = <0x60000000 0x10000>;
            bus-width = <4>;
            fifo-depth = <256>;
            interrupt-parent = <&{/soc/interrupt-controller@c000000}>;
            interrupts = <2>;
            max-frequency = <50000000>;
            cap-sd-highspeed;
            cap-mmc-highspeed;
            no-sdio;
        };
```

The startup error is as follows：
```
RISC-V 64, Boot ROM V3.6
Cannot access SD: Timeout
Cannot mount SD: Disk not ready
```
How can I fix this

NiborReb, 2025-05-12:
Did you make it run? 
Can you send the sdc.xdc constraints file?

NiborReb, 2025-06-02:
Hi, everytime I start the boot process the vcu118 fails at one point. Does anyone here have any idea?

[image]

[image]
[image]
[image]

elialaz, 2025-06-14:
Hi @NiborReb,

can you share your design? I'm trying the same thing but I'm having issue with the board setup.
Regarding the error you are encountering are you sure your SD card is properly formatted and working?

Thanks

NiborReb, 2025-06-17:
My design looks similar to the design above. My SD card was already working with the genesys2 board, so I'm not sure if the customizations are necessary. 
For the board setup, after selecting the vcu118 in the project settings, you can either run the auto connect so you don't need constraints for it, or customize the constraints to the pins in the datasheet, or just copy from other repos.

elialaz, 2025-06-17:
> I'll try reformatting the sd card and give a update.

Do it different times, even with OpenPiton the vcu118 with the Pmod sd card is still bugged as hell