eugene-tarassov/vivado-risc-v #248: Question about adding a GPIO

cassebas, 2024-09-10:
Hi Eugene,
Sorry, this may be a vague question.

I'd like to add a GPIO to the system. I know how to add an IO-block and add logic that interprets 'registers' that are communicated through the AXI-lite stream. But that's not what I'm after. My understanding of the IO-block is that it's memory mapped IO. The software writes values to a dedicated memory location and the IO-block sends the data to the AXI-slave. I'm guessing this kind of GPIO could have a high latency.

Another problem might be that the memory mapped IO is cached in the L1 cache? Or is this not the case?

I have seen a repository ```rocket-chip-blocks``` that can be included in the rocket-chip as a generator. I'm not sure whether this would be any better. Ideally I'd like to have a register within the processor that directly controls an output pin.

Do you have any ideas?
Thnx

eugene-tarassov, 2024-09-10:
> I'm guessing this kind of GPIO could have a high latency.

Yes, AXI bus has significant latency. AXI buses are pipelined to allow higher clock frequencies. The downside of this is higher latency.

> Another problem might be that the memory mapped IO is cached in the L1 cache?

No, caching is disabled for IO address range.

> I have seen a repository rocket-chip-blocks that can be included in the rocket-chip as a generator.

I'm not familiar with rocket-chip-blocks. In theory, they should have same latency as Vivado IP blocks, because they are connected to the same IO bus as memory mapped registers.

> Ideally I'd like to have a register within the processor that directly controls an output pin.

I think it can be done by creating a "Rocket Custom Coprocessor" (RoCC). I suspect it would be a lot of work.

eugene-tarassov, 2024-10-09:
> Would this latency be constant though, for multiple GPIO events?

No. AXI consists of 5 asynchronous buses, each bus has registers, FIFOs, switches and arbitration logic, the total latency depends on a lot of factors.

closed 2024-10-12 by cassebas