eugene-tarassov/vivado-risc-v #249: Question about the MEM_AXI4 to DDR communication

cassebas, 2024-10-09:
Hi Eugene,
I have made an extension to a 32-bit Rocket32s1 core, which monitors all AXI read requests from the core to main memory. My extension looks for specific addresses and outputs these read requests through an extra UART to my laptop. Below is a snippet of an example trace.

[example_trace_dram_monitor.txt](https://github.com/user-attachments/files/17310728/example_trace_dram_monitor.txt)

My question is about the AXI ARID, which is the fourth column in the trace. I recognize the requests in the complete trace, first I see the bootrom at work. It's copying the application binary from alt_mem to 0x80000000.

I think that when my application binary is running, I see that the ARID goes from 0x0 to 0x2. I wonder if this is indeed the case, and why it changes from 0x0 to 0x2 when the application is run? Is this done somehow by rocket-chip, or is it just a coincidence that I'm seeing the change after the bootrom copy action?

Thanks again!

eugene-tarassov, 2024-10-09:
ARID identifies bus master starting a read transaction. AXI enforces transaction ordering for same master, but can reorder transactions for different masters. You have to use ARID and RID to match replies to requests. AXI address and data busses are completely asynchronous, can reorder messages and allow multiple pending transactions. Basically, AXI bus works like 5 FIFOs per ID. I would expect each core and cache in RocketChip to have different IDs, but I'm not sure how exactly it is implemented.

cassebas, 2024-10-12:
Ok, thnx for the info.

closed 2024-10-12 by cassebas