eugene-tarassov/vivado-risc-v #250: Difference between bscan2jtag.vhdl and related IP from Vivado catalog

MaistoV, 2024-10-11:
As far as I understand, [bscan2jtag.vhdl](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/vhdl-wrapper/src/net/largest/riscv/vhdl/bscan2jtag.vhdl) converts from raw Xilinx BSCAN to plain JTAG.

On the other hand, Xilinx IP catalog features a very similar IP, namely BSCAN to JTAG Converter ([PG365](https://docs.amd.com/r/en-US/pg365-bscan-to-jtag-converter/Overview)), with only one configuration property (`CONFIG.enable_tck_bufg`). Such IP is only available in version 1.0, dated 2019.

Therefore, I have the following doubts:
- Is there any difference between the implementation in this repo and the one in the IP catalog? 
- Could they be interchanged in a basic "BSCAN to JTAG"-enabled design?
- Which one would you advice to use in which context?

Thank you for the help and availability.

eugene-tarassov, 2024-10-11:
Vivado BSCAN to JTAG IP was forked from the open source version a while ago. Since then, I did few minor improvements in the open source version, but Vivado version works fine too. 

> Which one would you advice to use in which context?

Normally BSCAN to JTAG IP is used together with Debug Bridge IP, but the open source version can also be used with BSCAN primitive, without Debug Bridge. In other words, bscanid_en signal is optional in the open source version.

MaistoV, 2024-10-14:
Thank you for the quick and clear response. 

I'll close this issue.

closed 2024-10-14 by MaistoV