eugene-tarassov/vivado-risc-v #251: Performance of rocket-chip

cassebas, 2024-10-27:
Hi Eugene,
I'm not sure if you can help me, but I'll try anyway. I'm running performance evaluations with a 32-bit Rocket-chip single core. I'm using benchmarks from the TacleBench benchmark suite, running them bare metal.

I have generated a rocket32s1 core from your repository, simply using the latest commit. (Without any modifications.) Then I run a TacleBench benchmark using the xsdb debugger, much like in your helloworld example.

In the benchmark code, I have added the cycle counter as a performance counter. As an example I use ```dijkstra```. Right before starting the benchmark, I save the number of cycles and right after running the benchmark I do this again. The difference is the number of cycles spent in the ```dijkstra_main``` function.

I do expect the number of cycles to vary somewhat, because of the DRAM, which I guess does not always give the same performance. It runs in a different clock domain compared to the core. But the strange thing is, sometimes after a reset (I push the button), the number of cycles is way off and it stays way off until I push the reset button again. In attached file you see some of the output, where each time I hit the reset button, the core starts again.

[taclebench-dijkstra-performance-rocket32s1.txt](https://github.com/user-attachments/files/17535114/taclebench-dijkstra-performance-rocket32s1.txt)

Do you have any clue as to why sometimes, the performance is worse? It seems that once in a while, the Rocket core wakes up in a bad mood or something.

Any ideas much appreciated!
thnx

eugene-tarassov, 2024-10-27:
It looks like the test reads some uninitialized memory location.
The reset causes random changes of DRAM data.
This can cause random variations in the test behavior.

eugene-tarassov, 2024-10-30:
As far as I know, flush/invalidate is not supported because it is not needed - RocketChip implements cache coherent interconnect.

eugene-tarassov, 2024-12-16:
XADC is enabled by default. It is programmed to turn on fan at 60C, turn off at 50C. The fan is not active because the temperature does not reach 60C. To see and edit default XADC setting, double click on the XADC block in Vivado. When `device_temp` changes, the DDR controller re-runs calibration, it uses test patters to find actual delay on DDR data lines to set optimal phase shift between data and clock. XADC does not require a driver, unless you want to adjust setting at run time, or to check current temperature and voltages.

cassebas, 2024-12-18:
Thank you. Indeed I can adjust the fan temperature, and I see my fan go on and off.

About this calibration, could it be that this is run after a reset? I can't image that this calibration runs during the processor's activity?

eugene-tarassov, 2024-12-18:
The re-calibration (aka temperature monitor calibration and dynamic calibration) can run during the processor's activity. According to the docs, it is done during idle cycles, and does not impact the performance of the memory interface. It is not same as initial calibration at power up.