eugene-tarassov/vivado-risc-v #253: Is 6-core possible?

binghe, 2024-11-01:
Hello,

I wonder if it's possible for your project to support a configuration of 6-core 64-bit RISC-V? My FPGA board (genesys2) has about 20k LUTs. 4 cores (64-bit RISC-V) take about 13k LUTs, while 8 cores seem impossible (without extra optimization).

Of course, I could just try it without asking first, but I'm still curious if there is any design limitations that the number of cores must be pow 2 values (1,2,4,8,16,...).

Chun TIAN

eugene-tarassov, 2024-11-01:
I have not tried it myself, but I believe RocketChip allows any number of cores.

binghe, 2024-11-01:
> I have not tried it myself, but I believe RocketChip allows any number of cores.

Thank you. I will have it a try and report back.

binghe, 2024-11-02:
It seems that 6 cores are indeed allowed by RocketClisp, but Digilent Genesys2 is still not big enough for 6 cores: when LUT utilization is very high (~95%), the implementation process (at route design) may fail. I give up on trying to generate more than 4 cores.