eugene-tarassov/vivado-risc-v #257: Issues with NFS Boot on Xilinx Alveo U250 - SFP Port Not Lighting Up & DHCP

Xu-feng-feng, 2024-11-29:
Hi, Eugene
Thank you for the open-source project. Could anyone provide insight into the possible causes of this issue or suggest a solution?
I have encountered an issue while attempting to boot a system on a Xilinx Alveo U250 using NFS. After generating the bitstream and downloading it to the XCU250, I noticed that the SFP port connected to the router does not light up. 

I followed the instructions in [[this issue thread](https://github.com/eugene-tarassov/vivado-risc-v/issues/62)](https://github.com/eugene-tarassov/vivado-risc-v/issues/62), and ran the following command:

```bash
make BOARD=u250 CONFIG= rocket64b2 \
ROOTFS_URL=192.168.1.100:/home/nfsroot \
HW_SERVER_ADDR=localhost:3121 JTAG_BOOT=1 jtag-boot
```

I’m using Vivado version 2023.2 for this project.
However, the process stops at:

```
loop: module loaded
```

The full boot log looks as follows:

```
RISC-V 64, Boot ROM V3.8

OpenSBI v1.4
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : freechips,rocketchip-vivado
Platform Features         : medeleg
Platform HART Count       : 2
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 1250000Hz
Platform Console Device   : axi-uart
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Platform Suspend Device   : ---
Platform CPPC Device      : ---
Firmware Base             : 0x80000000
Firmware Size             : 192 KB
Firmware RW Offset        : 0x20000
Firmware RW Size          : 64 KB
Firmware Heap Offset      : 0x27000
Firmware Heap Size        : 36 KB (total), 2 KB (reserved), 9 KB (used), 24 KB (free)
Firmware Scratch Size     : 4096 B (total), 320 B (used), 3776 B (free)
Runtime SBI Version       : 2.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff M: (I,R,W) S/U: ()
Domain0 Region01          : 0x0000000080020000-0x000000008002ffff M: (R,W) S/U: ()
Domain0 Region02          : 0x0000000080000000-0x000000008001ffff M: (R,X) S/U: ()
Domain0 Region03          : 0x000000000c000000-0x000000000fffffff M: (I,R,W) S/U: (R,W)
Domain0 Region04          : 0x0000000000000000-0xffffffffffffffff M: () S/U: (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000000010080
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes
Domain0 SysSuspend        : yes

Boot HART ID              : 0
[... 133 lines cut ...]
[    2.481648] 60010000.uart: ttyAU0 at MMIO 0x60010000 (irq = 12, base_baud = 0) is a axi-uart
[    2.490822] printk: legacy console [ttyAU0] enabled
[    2.490822] printk: legacy console [ttyAU0] enabled
[    2.501036] printk: legacy bootconsole [axi_uart0] disabled
[    2.501036] printk: legacy bootconsole [axi_uart0] disabled
[    2.603599] loop: module loaded

--- exit ---

```

After debugging, I found that the issue is related to a DHCP problem. The system seems unable to obtain an IP via DHCP, resulting in a timeout and the failure to mount the root filesystem via NFS. The issue logs show the following:
It seems like the system is unable to obtain an IP via DHCP, resulting in a timeout and the failure to mount the root filesystem via NFS.
```

[    2.602331] loop: module loaded
[    2.656500] usbcore: registered new interface driver uas
...
[    2.924755] Sending DHCP requests ...... timed out!
[   79.072854] IP-Config: Retrying forever (NFS root)...
[   79.105593] Sending DHCP requests ...... timed out!
[  166.792537] IP-Config: Retrying forever (NFS root)...
```

Could anyone provide insight into the possible causes of this issue or suggest a solution?

Xu-feng-feng, 2024-11-29:
I also tried running this command:

```bash
make BOARD=u250 CONFIG=rocket64b2 ROOTFS_URL=192.168.1.100:/sda2/nfsroot  HW_SERVER_ADDR=localhost:3121 ROOTFS=NFS JTAG_BOOT=1 jtag-boot
```

And the output was as follows:

```
RISC-V 64, Boot ROM V3.8

OpenSBI v1.4
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : freechips,rocketchip-vivado
Platform Features         : medeleg
Platform HART Count       : 2
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 1250000Hz
Platform Console Device   : axi-uart
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Platform Suspend Device   : ---
Platform CPPC Device      : ---
Firmware Base             : 0x80000000
Firmware Size             : 192 KB
Firmware RW Offset        : 0x20000
Firmware RW Size          : 64 KB
Firmware Heap Offset      : 0x27000
Firmware Heap Size        : 36 KB (total), 2 KB (reserved), 9 KB (used), 24 KB (free)
Firmware Scratch Size     : 4096 B (total), 320 B (used), 3776 B (free)
Runtime SBI Version       : 2.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff M: (I,R,W) S/U: ()
Domain0 Region01          : 0x0000000080020000-0x000000008002ffff M: (R,W) S/U: ()
Domain0 Region02          : 0x0000000080000000-0x000000008001ffff M: (R,X) S/U: ()
Domain0 Region03          : 0x000000000c000000-0x000000000fffffff M: (I,R,W) S/U: (R,W)
Domain0 Region04          : 0x0000000000000000-0xffffffffffffffff M: () S/U: (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000000010080
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes
Domain0 SysSuspend        : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART Priv Version    : v1.12
Boot HART Base ISA        : rv64imafdcx
Boot HART ISA Extensions  : none
Boot HART PMP Count       : 8
Boot HART PMP Granularity : 2 bits
Boot HART PMP Address Bits: 32
Boot HART MHPM Info       : 0 (0x00000000)
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

U-Boot 2022.01-dirty (Nov 29 2024 - 13:53:36 +0800)

CPU:   rv64imafdczicsr_zifencei_zihpm_xrocket
Model: freechips,rocketchip-vivado
DRAM:  14 GiB
MMC:   
Loading Environment from nowhere... OK
In:    uart@60010000
Out:   uart@60010000
Err:   uart@60010000
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
No MMC device available
No ethernet found.
No ethernet found.
=> 
```

It seems that the system still has trouble detecting the ethernet interface. Could you provide any advice on how to proceed?

---

eugene-tarassov, 2024-11-29:
The first command is the right one. The problem appears to be either wrong router or wrong Ethernet cable. You need to use 10G SFP+ router and connect it to the first lane of QSFP0 connector using  40G QSFP+ to 4 x 10G SFP+ Breakout Cable.

The message "No ethernet found" is coming from U-Boot, not from the OS. Ethernet in U-Boot is not needed, not supported and not used in this workflow.

Xu-feng-feng, 2024-11-30:
Thank you for your reply and suggestion. I followed your advice and used a 10G SFP+ router, connecting it to the first lane of the QSFP0 connector with a 40G QSFP+ to 4 x 10G SFP+ Breakout Cable.

However, after generating the bitstream for `rocket64b2` and downloading it to the XCU250, I noticed that the SFP port light does not turn on. When using a different bitstream (from [this source](https://github.com/alexforencich/verilog-ethernet/tree/master/example/Alveo/fpga_25g/fpga_AU250_10g)), the SFP port light does turn on, and the router interface shows that the SFP port is connected. But with the `rocket64b2` bitstream, the SFP port still shows as not connected.

I’m unsure what might be causing this discrepancy. Could there be something specific to the `rocket64b2` configuration that affects the SFP port's behavior?

eugene-tarassov, 2024-11-30:
In this repo, ethernet uses GT channel X1Y44, see [board/u250/ethernet-u250.tcl](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/board/u250/ethernet-u250.tcl), which matches docs for U250. However, that other example looks like it uses channel X1Y48 for the first lane of the QSFP0 connector. I'm not sure which one is correct, I don't have U250 board, I run tests on a different Alveo board.

Xu-feng-feng, 2024-11-30:
Hi Eugene,

Thank you for your guidance. I understand that the "No ethernet found" message is coming from U-Boot, not the OS. However, I am currently facing this issue where U-Boot is unable to detect Ethernet. Could you provide any insights on how to resolve this?

I have already ensured that I am using the correct 10G SFP+ router and connecting it properly via the 40G QSFP+ to 4 x 10G SFP+ Breakout Cable as per your recommendation.

Is there any additional configuration I need to check or set in U-Boot to enable Ethernet functionality, or could this be related to the specific hardware configuration for the rocket64b2 bitstream?

Any advice would be greatly appreciated!

eugene-tarassov, 2024-11-30:
This ethernet controller is not supported by U-Boot.

Xu-feng-feng, 2024-11-30:
I noticed that after running the `vivado_riscv64_defconfig` configuration, the `CONFIG_BOOTCOMMAND="booti ${kernel_addr_r} - ${fdt_addr}"` parameter was missing. After adding this parameter, the kernel successfully boots. However, during the boot process, the kernel stalls at the following log output:

```
RISC-V 64, Boot ROM V3.8

OpenSBI v1.4
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : freechips,rocketchip-vivado
Platform Features         : medeleg
Platform HART Count       : 2
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 1250000Hz
Platform Console Device   : axi-uart
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Platform Suspend Device   : ---
Platform CPPC Device      : ---
Firmware Base             : 0x80000000
Firmware Size             : 192 KB
Firmware RW Offset        : 0x20000
Firmware RW Size          : 64 KB
Firmware Heap Offset      : 0x27000
Firmware Heap Size        : 36 KB (total), 2 KB (reserved), 9 KB (used), 24 KB (free)
Firmware Scratch Size     : 4096 B (total), 320 B (used), 3776 B (free)
Runtime SBI Version       : 2.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff M: (I,R,W) S/U: ()
Domain0 Region01          : 0x0000000080020000-0x000000008002ffff M: (R,W) S/U: ()
Domain0 Region02          : 0x0000000080000000-0x000000008001ffff M: (R,X) S/U: ()
Domain0 Region03          : 0x000000000c000000-0x000000000fffffff M: (I,R,W) S/U: (R,W)
Domain0 Region04          : 0x0000000000000000-0xffffffffffffffff M: () S/U: (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000000010080
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes
Domain0 SysSuspend        : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART Priv Version    : v1.12
Boot HART Base ISA        : rv64imafdcx
Boot HART ISA Extensions  : none
Boot HART PMP Count       : 8
Boot HART PMP Granularity : 2 bits
Boot HART PMP Address Bits: 32
Boot HART MHPM Info       : 0 (0x00000000)
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

U-Boot 2022.01-dirty (Nov 30 2024 - 23:13:38 +0800)

CPU:   rv64imafdczicsr_zifencei_zihpm_xrocket
Model: freechips,rocketchip-vivado
DRAM:  14 GiB
MMC:   
Loading Environment from nowhere... OK
In:    uart@60010000
[... 107 lines cut ...]
[    1.099756] Unpacking initramfs...
[    1.117666] workingset: timestamp_bits=62 max_order=22 bucket_order=0
[    1.127658] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    1.138525] NFS: Registering the id_resolver key type
[    1.144034] Key type id_resolver registered
[    1.148783] Key type id_legacy registered
[    1.153241] nfs4filelayout_init: NFSv4 File Layout Driver Registering...
[    1.160529] nfs4flexfilelayout_init: NFSv4 Flexfile Layout Driver Registering...
[    1.171631] 9p: Installing v9fs 9p2000 file system support
[    1.180168] NET: Registered PF_ALG protocol family
[    1.186120] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 252)
[    1.193729] io scheduler mq-deadline registered
[    1.198620] io scheduler kyber registered
[    1.217136] riscv-plic c000000.interrupt-controller: mapped 8 interrupts with 2 handlers for 4 contexts.
[    2.229500] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    2.262696] 60010000.uart: ttyAU0 at MMIO 0x60010000 (irq = 12, base_baud = 0) is a axi-uart
[    2.271934] printk: legacy console [ttyAU0] enabled
[    2.271934] printk: legacy console [ttyAU0] enabled
[    2.281780] printk: legacy bootconsole [axi_uart0] disabled
[    2.281780] printk: legacy bootconsole [axi_uart0] disabled
[    2.382572] loop: module loaded

```

It seems that this issue may be related to the serial console output, possibly involving serial device drivers or console configuration. I have already verified the boot arguments, and the `console=ttyAU0,115200` setting is correct, but the issue persists.

eugene-tarassov, 2024-11-30:
To get right CONFIG_BOOTCOMMAND, use make arguments `ROOTFS=NFS  ROOTFS_URL=<your NFS share>`, see Makefile for details.
Also, see my comment https://github.com/eugene-tarassov/vivado-risc-v/issues/257#issuecomment-2508882728 about possible fix for the ethernet link.

Xu-feng-feng, 2024-12-01:
[image]
[image]

Thanks for the suggestions. I reviewed the configuration, and I found that there is no X1Y48 channel, so I’ve updated the script to match the same GT channel (X1Y44) as per your comment. However, the issue still persists.

I am currently trying a few other GT channels as well, but the ethernet link is still not working. Could you provide more details on what could be causing the problem? I’m using a different Alveo board, and everything seems to be set up according to your instructions, but the ethernet link is still not functioning properly.

eugene-tarassov, 2024-12-02:
In your screenshot, MGTREFCLK0 is selected. It is supposed to be MGTREFCLK1.
MGTREFCLK0  is not connected in this design. If you want to use it, you need to make changed in multiple design files.

Xu-feng-feng, 2024-12-02:
[image]

Thanks for the clarification. I see that the original design connects to MGTREFCLK1, but after downloading the bitstream, the SFP port still shows as not connected.

Could you please guide me on what specific changes I need to make to switch the reference clock to MGTREFCLK1 and properly configure the SFP port? I’m happy to provide more details about my setup if needed.

eugene-tarassov, 2024-12-02:
> Could you please guide me on what specific changes I need to make to switch the reference clock to MGTREFCLK1

MGTREFCLK1 is what you get if you do a clean build without any modifications.
As far as I know, it worked for others. I don't know why it does not work for you.
One possibility is you connected wrong channel to the router.
The board has 8 10G channels, only one (first channel of QSFP0) is supported in this design.

Xu-feng-feng, 2024-12-02:
Hi, Eugene

Thanks for the guidance! I made some changes to the config parameters like `CONFIG_ROOT_NFS`, and I’m really happy to finally get into the system. However, I’m encountering an issue with garbled output in the terminal. The system boots successfully, but the log messages are displaying strange characters, like this:

```
[    2.445248] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    2.480273] 60010000.uart: ttyAU0 at MMIO 0x60010000 (irq = 12, base_baud = 0) is a axi-uart
[    2.489461] printk: legacy console [ttyAU0] enabled
[    2.489461] printk: legacy console [ttyAU0] enabled
[    2.499688] printk: legacy bootconsole [axi_uart0] disabled
[    2.499688] printk: legacy bootconsole [axi_uart0] disabled
[    2.602712] loop: module loaded
[    2.657028] usbcore: registered new interface driver uas
[    2.661600] usbcore: registered new interface driver usb-storage
[    2.670650] mousedev: PS/2 mouse device common for all mice
[    2.681594] usbcore: registered new interface driver usbhid
[    2.686374] usbhid: USB HID core driver
[    2.691285] riscv-pmu-sbi: SBI PMU extension is available
[    2.696091] riscv-pmu-sbi: 16 firmware and 2 hardware counters
[    2.701283] riscv-pmu-sbi: Perf sampling/filtering is not supported as sscof extension is not available
[    2.717167] NET: Registered PF_INET6 protocol family
[    2.737036] Segment Routing with IPv6
[    2.740596] In-situ OAM (IOAM) with IPv6
[    2.744320] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    2.757261] NET: Registered PF_PACKET protocol family
[    2.762212] 9pnet: Installing 9P2000 support
[    2.767090] Key type dns_resolver registered
[    2.874448] Timer migration: 1 hierarchy levels; 8 children per group; 1 crossnode level
[    2.925372] Sending DHCP requests ., OK
[    3.468212] IP-Config: Got DHCP answer from 192.168.1.1, my address is 192.168.1.3
[    3.476058] IP-Config: Complete:
[    3.478582]      device=eth0, hwaddr=00:0a:35:00:00:03, ipaddr=192.168.1.3, mask=255.255.255.0, gw=192.168.1.1
[    3.488796]      host=192.168.1.3, domain=, nis-domain=(none)
[    3.494290]      bootserver=0.0.0.0, rootserver=192.168.1.100, rootpath=
[    3.494409]      nameserver0=192.168.1.1
[    3.507075] clk: Disabling unused clocks
[    3.740676] VFS: Mounted root (nfs4 filesystem) on device 0:17.
[    3.748984] devtmpfs: mounted
[    3.781097] Freeing unused kernel image (initmem) memory: 2180K
[    3.786477] Run /sbin/init as init process
[    9.599996] systemd[1]: System time before build time, advancing clock.
[   10.379048] systemd[1]: systemd 255.4-1+b1 running in system mode (+PAM +AUDIT +SELINUX +APPARMOR +IMA +SMACK +SECCOMP +GCRYPT -GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY +P11KIT +QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD -BPF_FRAMEWORK -XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified)
[   10.410946] systemd[1]: Detected architecture riscv64.

Welcome to ␛[1mDebian GNU/Linux trixie/sid␛[0m!

[   10.537673] systemd[1]: Hostname set to <debian>.
[   24.722514] systemd[1]: Queued start job for default target graphical.target.
[   24.975636] systemd[1]: Created slice system-getty.slice - Slice /system/getty.
[␛[0;32m  OK  ␛[0m] Created slice ␛[0;1;39msystem-getty.slice␛[0m - Slice /system/getty.
[   25.017320] systemd[1]: Created slice system-modprobe.slice - Slice /system/modprobe.
[␛[0;32m  OK  ␛[0m] Created slice ␛[0;1;39msystem-modprobe.slice␛[0m - Slice /system/modprobe.
[   25.057446] systemd[1]: Created slice system-serial\x2dgetty.slice - Slice /system/serial-getty.
[␛[0;32m  OK  ␛[0m] Created slice ␛[0;1;39msystem-serial\x2dget…slice␛[0m - Slice /system/serial-getty.
[   25.093150] systemd[1]: Created slice user.slice - User and Session Slice.
[␛[0;32m  OK  ␛[0m] Created slice ␛[0;1;39muser.slice␛[0m - User and Session Slice.
[   25.117024] systemd[1]: Started systemd-ask-password-console.path - Dispatch Password Requests to Console Directory Watch.
[␛[0;32m  OK  ␛[0m] Started ␛[0;1;39msystemd-ask-password-conso…␛[0mequests to Console Directory Watch.
[   25.146940] systemd[1]: Started systemd-ask-password-wall.path - Forward Password Requests to Wall Directory Watch.
[␛[0;32m  OK  ␛[0m] Started ␛[0;1;39msystemd-ask-password-wall.…␛[0md Requests to Wall Directory Watch.
[   25.183205] systemd[1]: Set up automount proc-sys-fs-binfmt_misc.automount - Arbitrary Executable File Formats File System Automount Point.
[␛[0;32m  OK  ␛[0m] Set up automount ␛[0;1;39mproc-sys-fs-binfm…␛[0mormats File System Automount Point.
[   25.209767] systemd[1]: Expecting device dev-ttyAU0.device - /dev/ttyAU0...
         Expecting device ␛[0;1;39mdev-ttyAU0.device␛[0m - /dev/ttyAU0...
[   25.227565] systemd[1]: Reached target cryptsetup.target - Local Encrypted Volumes.
[␛[0;32m  OK  ␛[0m] Reached target ␛[0;1;39mcryptsetup.target␛[0m - Local Encrypted Volumes.
[   25.248957] systemd[1]: Reached target integritysetup.target - Local Integrity Protected Volumes.
[␛[0;32m  OK  ␛[0m] Reached target ␛[0;1;39mintegritysetup.targ…␛[0m Local Integrity Protected Volumes.
[   25.273218] systemd[1]: Reached target paths.target - Path Units.
[␛[0;32m  OK  ␛[0m] Reached target ␛[0;1;39mpaths.target␛[0m - Path Units.
[... 15 lines cut ...]
[␛[0;32m  OK  ␛[0m] Listening on ␛[0;1;39msystemd-journald.socket␛[0m - Journal Socket.
[... cut ...]
[   26.450644] systemd[1]: Starting modprobe@dm_mod.service - Load Kernel Module dm_mod...
         Starting ␛[0;1;39mmodprobe@dm_mod.service␛[0m - Load Kernel Module dm_mod...
[   26.690465] systemd[1]: Starting modprobe@drm.service - Load Kernel Module drm...
         Starting ␛[0;1;39mmodprobe@drm.service␛[0m - Load Kernel Module drm...

```
It seems like some control characters are being displayed, which might be related to terminal settings or console configuration. 

Thanks again for your support!

eugene-tarassov, 2024-12-02:
> It seems like some control characters are being displayed

If you mean characters like `␛[0;` - those are ANSI escape codes. You need to use a console application with ANSI terminal emulation, like, for example, Tera Term.

Xu-feng-feng, 2024-12-02:
[image]
Thanks for your continued suggestion! I’ve downloaded and started using PuTTY, and the display is now showing correctly.I ran the RISC-V program and encountered the following error. I am currently trying to resolve it.

tahir2018, 2025-02-20:
Hi Xu Wenfeng, you mentioned in your comment above
 #Thanks for the guidance! I made some changes to the config parameters like CONFIG_ROOT_NFS
Can you please share what changes you make to config parameters? I am using U280 and stuck at same DHCP request timed out stage.

[  321.416415] IP-Config: Retrying forever (NFS root)...
[  321.454582] Sending DHCP requests ...... timed out!
[  408.766383] IP-Config: Retrying forever (NFS root)...
[  408.804457] Sending DHCP requests ...... timed out!

Thank you for your response.