eugene-tarassov/vivado-risc-v #263: Generating genesys2-riscv.mcs on WSL?

matsbror, 2025-01-27:
As the title says, I am using Ubuntu 20.04 on WSL2 and the idea is to program the Genesys2 board from Vivado on Windows. I cannot find the file: workspace/rocket64b2/genesys2-riscv.mcs. I thought it would be generated at the stage `make CONFIG=rocket64b4 BOARD=genesys2 vivado-tcl` but apparently not.

Do I need to run vivado in Linux to make it work?

eugene-tarassov, 2025-01-27:
You can run Vivado on Windows. `make ... vivado-tcl` creates a script that you can use on Windows to create Vivado project, then use Vivado GUI to generate bitstream and mcs files.

matsbror, 2025-01-27:
Thanks @eugene-tarassov , I did just that and I could generate the bitstream, but did not see how to generate the mcs file. I will look again tomorrow.

eugene-tarassov, 2025-01-27:
The error message does not make sense. I think the actual cause of the error is something simple, like wrong current directory or a typo in the file name.

matsbror, 2025-01-28:
I have an issue that Vivaod (which runs in Windows) seems to not store the resulting files correctly. I can run "Generate Bitstream", but there is no .bit-file generated and it does not remember it had done the Synthesis and Implementation. I see no errors.

Can it have something to do with Windows vs linux file system?

eugene-tarassov, 2025-01-28:
Windows has file path length limitation, 256 characters by default. If your vivado-risc-v is not a top level folder, it can cause problems.

matsbror, 2025-01-29:
In the end I resolved it by installing vivado also on WSL and generating bitstream there and using vivado on Windows to program the FPGA.

closed 2026-08-19 by matsbror