eugene-tarassov/vivado-risc-v #265: What is normal network speed?

matsbror, 2025-01-31:
I have a rocket64b4  on genesys2 board and finally got it working.

I have a question about network speed. It seems pretty slow to me, so I tested with speedtest-cli and got about 5.5 Mbit/s.
Is this what I should expect from an FPGA-based core?

eugene-tarassov, 2025-01-31:
Using iperf3 on genesys2 64b2, I got 28.5 Mbits/sec, which is about right for a CPU running at 100MHz.

matsbror, 2025-01-31:
I got this result when running iperf3 as server on the genesys2 board:
```
Connecting to host genesys2.uni.lux, port 7575
[  5] local 172.29.94.165 port 46796 connected to 10.186.109.12 port 7575
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  3.81 MBytes  31.9 Mbits/sec   28   22.6 KBytes
[  5]   1.00-2.00   sec  3.80 MBytes  31.9 Mbits/sec   24   35.4 KBytes
[  5]   2.00-3.00   sec  3.62 MBytes  30.4 Mbits/sec   24   32.5 KBytes
[  5]   3.00-4.00   sec  3.62 MBytes  30.4 Mbits/sec   15   33.9 KBytes
[  5]   4.00-5.00   sec  3.74 MBytes  31.4 Mbits/sec   23   31.1 KBytes
[  5]   5.00-6.00   sec  3.80 MBytes  31.9 Mbits/sec   21   25.5 KBytes
[  5]   6.00-7.00   sec  3.56 MBytes  29.9 Mbits/sec   35   24.0 KBytes
[  5]   7.00-8.00   sec  3.56 MBytes  29.9 Mbits/sec   28   19.8 KBytes
[  5]   8.00-9.00   sec  3.74 MBytes  31.4 Mbits/sec   30   19.8 KBytes
[  5]   9.00-10.00  sec  3.68 MBytes  30.9 Mbits/sec   28   24.0 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  36.9 MBytes  31.0 Mbits/sec  256             sender
[  5]   0.00-10.02  sec  36.6 MBytes  30.7 Mbits/sec                  receiver
```

closed 2026-08-19 by matsbror