eugene-tarassov/vivado-risc-v #266: migration U55C

vubinstar1219, 2025-02-28:
hello. I would like to migrate my u200 or u250 system to u55c. 

However, during the process, I have a question and would like to ask a question. Did you write and configure files such as ethernet-u200.v and internal modules yourself?

eugene-tarassov, 2025-03-01:
> Did you write and configure files such as ethernet-u200.v and internal modules yourself?

I was involved.

> I have a question and would like to ask a question. 

Sure, ask it.

vubinstar1219, 2025-03-07:
Could you elaborate on what the role of ethernet is? And what are the things to consider before porting to u55c? 

Sorry, I am new to FPGAs so I don't have enough knowledge.

vubinstar1219, 2025-03-12:
I have confirmed that there are two riscv cores on the current alveo u55c product through xsdb. 

However, I don't know much about the ethernet part, so I want to boot riscv linux in a different way. If there is another way, can you explain what principle it follows? 

Thank you for sharing the wonderful project and answering the issue.