eugene-tarassov/vivado-risc-v #268: Gemmini illegal instruction

ethan2339, 2025-03-05:
# Gemmini Illegal Instruction Error on Genesys 2 Board

I've implemented my Genesys 2 board with `CONFIG=Rocket64b1gem8` using the standard configuration from `rocket.scala` and I'm encountering issues when trying to execute Gemmini rocc tests. When running the Linux executables on genesys 2 from  `build/bareMetalC/ -linux ` compiled from the gemmini-rocc-tests in the vivado-risc-v directory, I'm either getting an illegal instruction error or no output at all. Almost every -linux executables are showing the same problem, with some of them displaying failure messages.

I've verified in Vivado that Gemmini is included in the design and confirmed the LUT usage. I've also checked the gemmini_params.h header file, compiled using the standard RISC-V and Chipyard toolchain, and tried both the image from the release page and one generated with `./mk-sd-image`, but with the same results.

My environment:
- Ubuntu 20.04 LTS
- Vivado 2024.2

I've attempted this multiple times and I'm not sure what I might be missing. I'm hoping to get some clues to resolve this issue, as I ultimately want to use Gemmini for ONNX model research but I'm currently stuck at this point.

Thank you to everyone in this community for your help.

[image]

[image]

[image]

[image]

hansa31, 2025-03-24:
This happens because of the rdcycle() instruction generation. In the tiled_matmul code there is a line counting CPU cycles. Comment this line and it should work. (rdcycle won't work because of previllages)

ethan2339, 2025-03-28:
@hansa31 Thank you so much ! It has solved my problem !

closed 2025-03-28 by ethan2339