eugene-tarassov/vivado-risc-v #269: Kernel compilation/assembler issues when attempting to create SD card image

bdswimley, 2025-03-08:
When I run "./mk-sd-card", I'm getting several errors related to 'unrecognized opcodes' during the build of the Linux kernel and drivers per the following build output:

bswimley@SwimleyLaptop:/mnt/d/vivado-risc-v$ ./mk-sd-card
make: 'debian-riscv64/initrd' is up to date.
make: 'debian-riscv64/rootfs.tar.gz' is up to date.
make -C u-boot CROSS_COMPILE=/usr/bin/riscv64-linux-gnu- BOARD=vivado_riscv64 vivado_riscv64_config
make[1]: Entering directory '/mnt/d/vivado-risc-v/u-boot'
#
# configuration written to .config
#
make[1]: Leaving directory '/mnt/d/vivado-risc-v/u-boot'
make -C u-boot \
  BOARD=vivado_riscv64 \
  CC=/usr/bin/riscv64-linux-gnu-gcc \
  CROSS_COMPILE=/usr/bin/riscv64-linux-gnu- \
  KCFLAGS='-O1 -gno-column-info' \
  u-boot-nodtb.bin
make[1]: Entering directory '/mnt/d/vivado-risc-v/u-boot'
scripts/kconfig/conf  --syncconfig Kconfig
  CFG     u-boot.cfg
  GEN     include/autoconf.mk.dep
  GEN     include/autoconf.mk
  UPD     include/generated/timestamp_autogenerated.h
  CC      arch/riscv/cpu/cpu.o
  AS      arch/riscv/cpu/mtrap.o
  AS      arch/riscv/cpu/start.o
arch/riscv/cpu/mtrap.S: Assembler messages:
arch/riscv/cpu/mtrap.S:65: Error: unrecognized opcode `csrr a0,scause', extension `zicsr' required
arch/riscv/cpu/mtrap.S:66: Error: unrecognized opcode `csrr a1,sepc', extension `zicsr' required
arch/riscv/cpu/mtrap.S:67: Error: unrecognized opcode `csrr a2,stval', extension `zicsr' required
arch/riscv/cpu/mtrap.S:70: Error: unrecognized opcode `csrw sepc,a0', extension `zicsr' required
make[2]: *** [scripts/Makefile.build:289: arch/riscv/cpu/mtrap.o] Error 1
make[2]: *** Waiting for unfinished jobs....
arch/riscv/cpu/start.S: Assembler messages:
arch/riscv/cpu/start.S:65: Error: unrecognized opcode `csrw stvec,t0', extension `zicsr' required
arch/riscv/cpu/start.S:72: Error: unrecognized opcode `csrw sie,zero', extension `zicsr' required
arch/riscv/cpu/start.S:346: Error: unrecognized opcode `csrw stvec,t0', extension `zicsr' required
make[2]: *** [scripts/Makefile.build:289: arch/riscv/cpu/start.o] Error 1
arch/riscv/cpu/cpu.c: Assembler messages:
arch/riscv/cpu/cpu.c:94: Error: unrecognized opcode `csrs sstatus,a5', extension `zicsr' required
arch/riscv/cpu/cpu.c:95: Error: unrecognized opcode `csrw 0x003,0', extension `zicsr' required
make[2]: *** [scripts/Makefile.build:254: arch/riscv/cpu/cpu.o] Error 1
make[1]: *** [Makefile:1810: arch/riscv/cpu] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC      drivers/timer/riscv_timer.o
...

Please note that I was able to generate the HDL for import into Vivado 2024.2 and then able to generate the bitstream successfully for the rocket64b1 on my Nexys-A7-100T board.  I then used the prebuilt SD card image to verify that it boots/runs successfully.

Thanks in advance for any pointers and suggestions.

Best Regards,
-Bryce

eugene-tarassov, 2025-03-09:
It works fine to me. The GCC version:
```
eugene@gate:~$ /usr/bin/riscv64-linux-gnu-gcc --version
riscv64-linux-gnu-gcc (Ubuntu 9.4.0-1ubuntu1~20.04) 9.4.0
```
Note that this repo is prepared to be used with Ubuntu 20.04 LTS.
If you are using any other Linux distribution, you will see problems.

bdswimley, 2025-03-09:
Thanks for the prompt response, Eugene.  My environment is indeed different - Ubuntu 24.04.2 LTS, running under WSL on a Win11 machine.  I'll look into installing Ubuntu 20.04 LTS..

Thanks again,
-Bryce