eugene-tarassov/vivado-risc-v #271: Debian boot stuck on MPSoC

ipapal, 2025-03-20:
Hello,

Thank you for providing this repository. I am in the process of migrating the Chipyard RISC-V project from the VCU118 to the ZCU106 board. The migration has been successful, and I have generated a working bitstream. However, I am facing issues when trying to boot Linux.

I created the SD card by following the commands in the ```README.md``` file (executing the ```mk-sd-card``` script). However, when I attempt to boot, the process gets stuck at the following output:

```
INIT  
CMD0  
CMD8  
ACMD41  
CMD58  
CMD16  
CMD18  
LOADING 0x01e00000B PAYLOAD  
LOADING  
BOOT  
```

Is it necessary to manually configure the SD card rather than using the pre-installed one? Additionally, I am not using configuration memory, I am loading the plain bitstream directly from the Hardware Manager. I don’t believe this should cause any issues. Furthermore, the SD card appears to be recognized correctly, as I receive the appropriate error when it is not plugged in.

Any insights would be greatly appreciated.

Kind regards

eugene-tarassov, 2025-03-20:
This repo has its own implementation of BootROM, UART, SD and Ethernet, and all related drivers. The Chipyard generated system cannot boot SD card from this repo, unless the Chipyard project is modified to use IO modules from this repo.

ipapal, 2025-03-22:
I understand, therefore the only way forward is to integrate our custom Scala code into your repository. I believe that changing the FPGA target device from the ones currently supported in this repo to the MPSoC should be feasible, as others have successfully ported this GitHub project to different FPGAs, such as the VCU118, based on previous discussions in the issues section.

closed 2025-04-02 by ipapal