eugene-tarassov/vivado-risc-v #272: rocket64w1 timing on genesys2

adambagley, 2025-03-23:
Hi @eugene-tarassov ,

Thank you so much for your repo here, it's really fantastic. I'm using it for a school project where I generated bitstreams for 8 different flavors for genesys2 board and comparing them against each other using various CPU benchmarking utilities.

Here are the 8 flavors I was able to fit on the genesys2:
rocket64b1
rocket64b2
rocket64b2gem4
rocket64b2l2
rocket64b4l2w
rocket64b4
rocket64w1
rocket64x1

(These flavors are too big and don't fit on genesys2):
rocket64b8
rocket64y1
rocket64z1

Everything going great so far except one minor thing--the rocket64w1 build for genesys2 fits and finishes routing and bitstream generation (and even appears to be perfectly functional on hardware), but it fails timing by  -1.6ns. I think it might just be a constraint issue. But curious if you've ever tried this permutation before (rocket64w1 on genesys2) and if you have any ideas of how to fix. Interestingly the bigger BOOM core rocket64x1 passes timing just fine on genesys2, which leads me to believe this is simply a constraint issue with the smaller BOOM.

I attached a screenshot of the worst path. I'm on Vivado 2024.2.
Thank you!
Adam

[image]

eugene-tarassov, 2025-03-24:
This file [board/rocket-freq](https://github.com/eugene-tarassov/vivado-risc-v/blob/master/board/rocket-freq) defines clock frequencies for various Rocket configurations. The default 100MHz for rocket64w1 appears a bit high. The supported values are (MHz): 160 125 100 80 62.5 50 40 31.25 25 20. Chose one that does not fail timing.

eugene-tarassov, 2025-03-25:
1. I don't know much about Gemmini, but I recall that others were able to get it working.
2. The L2 cache always on and works under-the-hood. The OS gets info about caches in the device tree. I don't know how the OS uses this info. The RocketChip maintains cache coherency on the hardware level, so the software does not need to do anything with the caches.

closed 2025-03-25 by adambagley

adambagley, 2025-04-08:
Hi @eugene-tarassov quick follow-up question,
What's the reasoning for the supported clock freq values being capped at 160MHz? I'm finding on my speedgrade3 Ultrascale+ board I can close timing easily at 160MHz on Rocket64b1 and I'm curious about trying to push the clock faster.
Thank you!
Adam

eugene-tarassov, 2025-04-08:
No such reason. You can use any frequency as log as the Clock Wizard can synthesize all clocks exactly as requested - without rounding.