eugene-tarassov/vivado-risc-v #275: Migration to MPSoC (ZCU106)

ipapal, 2025-04-02:
Hello,

I have started migrating your design to the ZCU106. However, when I try to boot, there is no output on the UART. I came across previous issues related to the VCU118, but I haven't had any luck resolving the problem. When using XSDB, the output appears as expected. Additional debug information from ILAs includes:
The dbg_bus on the DDR4 MIG outputs only "0000000...0".
The MAXI and SAXI ports of the SD card module do not show any transactions.
I am using a PMOD SD adapter. Apart from incorporating a DDR4 MIG, are there any other modifications I should make?

Thank you for your help!
Best regards

eugene-tarassov, 2025-04-04:
Most probably, this is related to the PMOD SD card adapter. Does U-Boot detects the SD CARD (MMC)? It should look like this:
```
U-Boot 2022.01-dirty (Apr 03 2025 - 18:33:50 -0700), Build: jenkins-tcf-jtag-test-nexys-linux-2024.2-60

CPU:   rv64imafdczicsr_zifencei_zihpm_xrocket
Model: freechips,rocketchip-vivado
DRAM:  512 MiB
MMC:   mmc0@60000000: 0
```

ipapal, 2025-04-04:
Yes, @eugene-tarassov it look's like it detects the SD CARD, it looks similar to your output: 
```
U-Boot 2022.01-dirty (Apr 04 2025 - 19:31:21 +0300)

CPU:   rv64imafdczicsr_zifencei_zihpm
Model: freechips,rocketchip-vivado
DRAM:  1 GiB
MMC:   mmc0@60000000: 0
```
Also, in the Vivado address editor, the addresses are the same with the vcu707 project. 
PS: I'm including in a file the full log file. 
[log_zcu106.txt](https://github.com/user-attachments/files/19609843/log_zcu106.txt)

eugene-tarassov, 2025-04-04:
One possible reason why SD works in U-Boot, but does not work in the Linux kernel is that Linux is using 4-bit SD data bus, while U-Boot uses 1-bit bus. If SD adapter is not properly wired to support 4-bit bus, the Linux boot fails.

ipapal, 2025-04-04:
Ok, thank you I will check it. I'm using the popular PMOD SD card adapter from [Digilent](https://digilent.com/reference/pmod/pmodmicrosd/reference-manual?redirect=1), which seems like it can operate in 1-bit and 4-bit mode, so maybe it can work without changing the SD card controller code.

ipapal, 2025-04-05:
An update @eugene-tarassov , I have checked also the arty a7 100t project, that uses the digilent PMOD SD adapter (I am using the micro SD version, not the full SD card but they have the same pins/interface). In this project, I saw that nothing changes in the SD card controller Verilog code, only the mapping in the constraints to the respective PMOD pins. I have checked carefully the connections in my project with the PMOD connector and I believe they are correct. However, the issue still remains the same. The only difference I was thinking it might affect the expected operation, is that the PMOD in zcu106 operates in 1.8V (instead of the 3.3V of the a7 100t), but it has an internal level shifter that translates the voltage levels.

ipapal, 2025-04-05:
I have figured it out, I have configured wrong the SD card controller (SD card detect). Thank you for your support !

closed 2025-04-05 by ipapal