eugene-tarassov/vivado-risc-v #277: JTAG Booting Error (Putty)

vubinstar1219, 2025-04-13:
Hello. I am in the process of porting RISC-V Linux using JTAG. I'm encountering an issue where the text appears garbled in Putty. Do you happen to know how to resolve this kind of problem?
The baud rate is set to 115200, and the same value is configured in Putty.
Are there any other possible causes for this issue?
I'm currently using riscv-linux-gcc version 9.2.0.

[image]

adambagley, 2025-04-18:
In my experience baud rate mismatch is typical cause of this, but if you're sure that's not the problem, you should confirm you also have 8N1 set as well (8 data bits, no parity bits, 1 stop bit). 
I would also suggest, simply to control the experiment, trying alternative serial port terminal apps besides Putty. If you're using Windows there's a good one called TeraTerm. If you have access to Linux some good ones are cutecom and minicom

vubinstar1219, 2025-04-18:
> In my experience baud rate mismatch is typical cause of this, but if you're sure that's not the problem, you should confirm you also have 8N1 set as well (8 data bits, no parity bits, 1 stop bit). I [...]

Thank you for your response.

I tried the methods suggested by @adambagley, but unfortunately, the text still appears garbled. I also tested writing the character "A" using a command like mwr 0x60010004 0x41, as mentioned in another GitHub repository, but the output was still distorted.

Is there any other possible solution you could suggest? Also, I noticed that the Rocket Core and the UART are mapped together in the device tree structure. Could the issue be originating from the Rocket Core itself, causing the corrupted output? I understood that data is passed to the UART through the MMIO port, but if that's a misunderstanding on my part, I’d appreciate any clarification or guidance.

vubinstar1219, 2025-04-29:
> In my experience baud rate mismatch is typical cause of this, but if you're sure that's not the problem, you should confirm you also have 8N1 set as well (8 data bits, no parity bits, 1 stop bit). I [...]

[image]

Hello,
I have resolved the UART issue. However, I am encountering a problem during the booting process, as shown in the attached image.
Do you have any advice or references that could help me address this issue?