eugene-tarassov/vivado-risc-v #278: license agreement

adambagley, 2025-04-14:
Hi @eugene-tarassov ,

I've been using your repo for a personal school project unrelated to my day job and as you know I did a couple pull requests for changes I wanted to contribute.

However I'm also looking to possibly experiment with some stuff from your repo for my day job and make some pull request contributions from that as well, but my employer requires there's an open source license in place in order for me to do that. Would you consider throwing up an open source license for your repo?

Thank you,
Adam

eugene-tarassov, 2025-04-14:
This is composite repo which includes submodules with different licenses, e.g. Linux kernel - GNU License, RocketChip - Berkeley + SiFive (Apache) License, SD card - GNU Lesser License, etc. I'm not sure what kind of license I should put in the root directory. Perhaps your employer cat tell what they expect to see.

adambagley, 2025-04-15:
So I did quite a bit of research on this.
Basically we need to consider what the licenses are for the submodules, and pick a root license that is compatible with all of them, even (especially) the strictest ones. And irrespective of my employer, after doing this research I think it's a very good idea to include a license just to be safe. So I took inventory of what the current submodule licenses are:

- rocket-chip:  Apache 2.0 from SiFive, plus its own Berkeley license, plus another Berkeley license for chisel-jtag
- u-boot: GPL 2.0+
- linux-stable: GPL 2.0
- verilog-ethernet: MIT
- sifive-cache: Apache 2.0
- testchipip: BSD-3
- riscvboom: BSD-3, plus Apache 2.0 from SiFive
- gemmini: its own Berkeley license
- opensbi: BSD-2
- coremark: its own license, plus Apache

From what I gather, GPL 2.0 is not compatible with Apache 2.0, but both (and all the others) are compatible with GPL 3.0. So I created a proposed LICENSE.md in a [pull request ](https://github.com/eugene-tarassov/vivado-risc-v/pull/279)that I think covers all the bases, if you want to take a peek when you get a chance.

eugene-tarassov, 2025-04-15:
I believe we should not be concerned about GPL 2.0 being not compatible with Apache 2.0, because compactivity is required for "libraries", not "submodules". In the context of GPL, a library refers to a collection of software functions and/or data designed to be easily linked with application programs. In this project, the Linux kernel is not linked with RoketChip or anything else, so it is not "included as a library". Same for U-Boot. I'd like to exclude GPL from the consideration, because it is a toxic license. For the common license, I'd look for compatibility with MIT, BSD and Apache.

adambagley, 2025-04-15:
Sounds good, I just updated the pull request with MIT license instead

closed 2025-04-15 by adambagley